SRA-VJTI/sra-board-hardware-design #3: Add Datasheets folder

udit7395, 2021-02-12:
For eg we use IC LM324 , Motor Driver XYZ, LDO, etc

ombhilare999, 2021-02-12:
Sra board hardware design repo already has a datasheet folder, I haven't added all the datasheets there, didn't want to increase the size of that repo.
@laukik-hase We should add all the datasheet links maybe in a readme.
https://github.com/SRA-VJTI/sra-board-hardware-design/tree/master/datasheets

VedantParanjape, 2021-02-12:
@ombhilare999 datasheets are hardly 100kb, won't increase the size much.

ombhilare999, 2021-02-13:
[image]
https://github.com/SRA-VJTI/sra-board-hardware-design/tree/master/datasheets
The current size of the datasheet folder, I guess all links in one readme serve the purpose.
@laukik-hase @VedantParanjape

VedantParanjape, 2021-02-13:
> [image]
> https://github.com/SRA-VJTI/sra-board-hardware-design/tree/master/datasheets
> The current size of the datasheet folder, I guess all links in one readme serve the purpose.
> @laukik-hase @VedantParanjape 
> 

Links can be dead at any point, it's better to have separate pdf

udit7395, 2021-02-16:
@VedantParanjape If a consensus has been reached, close the issue

closed 2021-02-16 by VedantParanjape