SRA-VJTI/sra-board-hardware-design #13: Add Pinout Image of SRA board and sensors

ombhilare999, 2022-02-17:
Something like this can be done for the boards and sensors:

Example 1:
[image]

Example 2:
[image]

ombhilare999, 2022-02-17:
@dhairyashah1 assign this task to some second-year student.

dhairyashah1, 2022-02-17:
@omsheladia i hope you can prepare the same. Get working on it.

rahulsingh701, 2022-02-23:
update @dhairyashah1 ?

dhairyashah1, 2022-02-23:
[image]
[image]
[image]

ombhilare999, 2022-02-24:
The first Image needs improvements, too many arrows crossing each other.
Check out this: https://hackaday.com/2021/08/03/tool-generates-interactive-pcb-diagrams-from-kicad/
Demo: https://blog.honzamrazek.cz/2021/06/making-nice-looking-and-interactive-diagrams-for-your-pcbs/

VedantParanjape, 2022-08-30:
@gautam-dev-maker remove inactive members from this project (@omsheladia )

omsheladia, 2022-08-30:
The tool suggested by Omkar didn't seem to work on my system then, however if crossing of arrows is the only concern then I have found something else that can do the work, @gautam-dev-maker  let me know if I can carry on with that.

dhairyashah1, 2022-08-30:
What is that something else? Links / demos are desired @omsheladia

VedantParanjape, 2022-08-30:
> The tool suggested by Omkar didn't seem to work on my system then, however if crossing of arrows is the only concern then I have found something else that can do the work, @gautam-dev-maker  let me know if I can carry on with that.

I never saw error logs or you trying to get that tool working.

omsheladia, 2022-08-30:
@dhairyashah1 To prevent the crossing of arrows, we can insert an image in any flowchart maker and label them with more organised arrows compared to the ones that I used in Canva previously.

omsheladia, 2022-08-30:
@VedantParanjape I talked about it personally to Omkar but we didn't reach towards any conclusion then.

ombhilare999, 2022-08-31:
@omsheladia you sure about this?
Didn't I tell you to explore the VM option?

gautam-dev-maker, 2022-08-31:
@omsheladia I guess I have listed steps to run linux in parallels VM (during Walle). In fact I installed it in your system if I remember correctly.

dhairyashah1, 2023-01-19:
@RajGupta17 @Greyless such pinouts are expected in the README.md's for SRA_Board_2023 and LSA, MPU respectively. Kindly incorporate these in your PRs for documentation.

dhairyashah1, 2023-04-25:
Closing this as labelled pinout images are already incorporated

closed 2023-04-25 by dhairyashah1