SRA-VJTI/sra-board-hardware-design #15: Replace LEDs and phototransistor on LSA with IR sensors.

gautam-dev-maker, 2022-09-14:
The current design of LSA is very prone to ambient light and often cause inconvenience in line following.
Possible solution is to test [TCRT-5000](https://robu.in/product/tcrt5000-reflective-ir-sensor-photoelectric-switch/?gclid=Cj0KCQjw94WZBhDtARIsAKxWG-88AqPIVCcqhdT3gfbs87LkVztaSn-T9nJmmQKPh13s6nakmClp1rsaAvKUEALw_wcB) for ambience interference. 

If it proves to be better than current design, we can easily modify our LSA design to include it.

dhairyashah1, 2022-11-01:
As discussed previously

- [ ] Try different IR based Transmitter Receiver Pairs ( PTH and SMD ) from various LAMI vendors on a perfboard 
- [ ] Test on the current SRA Board and check readings on ADC lines for line-following with and without a buffer amplifier IC
- [ ] If buffer amplifier is needed, find a dealer (online or offline) if HEX IC is available ( 5 -6 gates in one IC ) for the facility of more than 4 sensor pairs if desired
- [ ] Redesign LSA PCB prototype for the best available combinations
- [ ] Test PCB Prototype
- [ ] Final Order

Hard deadline - 5th November 

@marck3131 proceed with this issue and update ASAP

ChinmayLonkar, 2022-11-03:
@marck3131 Please post the perfboard testing results here ASAP

dhairyashah1, 2022-11-13:
ping @marck3131. Please post the results here as well.

marck3131, 2022-11-14:
5 IR sensors were tested on perfboard with resistors 4.7k E and 100 E , with gpio pin numbers 32, 34, 35, 36, 39
Tested Line following too with the same ...

dhairyashah1, 2022-11-14:
@marck3131 Please put the images, sensor placement and line-following demo video with new config here as well for reference.

ChinmayLonkar, 2023-01-13:
@dhairyashah1 we can close this issue as this has been implemented in the design.

dhairyashah1, 2023-01-13:
> @dhairyashah1 we can close this issue as this has been implemented in the design.

LGTM. Design PR #21 merged in master.

dhairyashah1, 2023-01-13:
Feature PR #21 merged in master

closed 2023-01-13 by dhairyashah1