SRA-VJTI/sra-board-hardware-design #16: Replacing extra Motor driver with Stepper Driver

dhairyashah1, 2022-10-29:
- To replace and add functionality to drive a stepper motor on SRA Board - PCB design
- Mention specifications of the driver used & models of stepper tested across
- Add Terminal blocks to connect the servo motor
- Pin headers/Pads/terminals for external supply (if desired to operate on higher rating (not from SRA board 5V/3V3))

Suggestion - 
Include a Switch to toggle between power supply from SRA Board <-> External supply connected to port/pin header/pad

CC: @ChinmayLonkar @Asc91
Share the Updates with Images of PCB and changes made so far here

Asc91, 2022-10-30:
[image]
[image]

This is initial version of SRA board with added stepper motor driver.
Driver used : [DRV8825](https://www.tme.eu/Document/1dd18faf1196df48619105e397146fdf/POLOLU-2133.pdf)
Motor Tested: Nema 17

https://user-images.githubusercontent.com/55803500/198893070-4ce25d15-41e1-474a-a37e-f0e8930bdfe7.mp4

Current 12V adapter is used to power the motor.

ChinmayLonkar, 2022-11-03:
@dhairyashah1 ,
I have created a pull request, Kindly review and let me know if there re any changes to be made

dhairyashah1, 2023-02-02:
@ChinmayLonkar is the current design compatible with housing either DRV8825 or A4988 without any other changes?

ChinmayLonkar, 2023-02-04:
Yes, we can use both of them without any changes.

closed 2023-03-24 by dhairyashah1