SRA-VJTI/sra-board-hardware-design #17: [ PCB Design ] Add Pin header in front of current MPU Port for direct MPU6050 plugin

dhairyashah1, 2022-10-29:
- The noise observed on the SDA and SCL lines of MPU6050 leads to an I2C Packet drop while self-balancing (motors running).
- This is related to interference between the BO Motor wires and the long MPU JST Cable.
- Shortening the length of I2C lines prevents this issue. Ref #14 
- This issue aims to add a direct pin header port for direct plug-play and operation of the MPU6050 on the SRA Board

Todo - 
- [ ] Update test results for direct plugin MPU near the current port
- [ ] Test the above thoroughly with self-balancing
- [ ] Make changes in the PCB Design
- [ ] Updated PCB Fabrication, Assembly
- [ ] Rigorous testing for self-balancing and making suitable changes ( like using MPU Offset) in the [firmware](https://github.com/SRA-VJTI/Wall-E/tree/master) for[ 3_MPU](https://github.com/SRA-VJTI/Wall-E/tree/master/3_MPU) and [8_self_balancing](https://github.com/SRA-VJTI/Wall-E/tree/master/8_self_balancing)

CC - @ChinmayLonkar @Premraj02

ChinmayLonkar, 2022-10-31:
@Premraj02 take up this task and post the updates before 2nd November.

ChinmayLonkar, 2022-11-03:
As per our testing pin header is not required as we can directly use the shielded cables which resolves the issue,
@Premraj02 you can put up the test results here for the test in which you connected the MPU near the current port.

dhairyashah1, 2022-11-13:
ping @Premraj02 .Put/link the test results so that I may close this issue.

Premraj02, 2022-11-20:
[This](https://github.com/SRA-VJTI/sra-board-hardware-design/issues/14#issuecomment-1321137060) test solved the noise issue. So no need to add pin header.

closed 2022-11-20 by dhairyashah1