SRA-VJTI/sra-board-hardware-design #18: Add web page for Board Hardware Documentation, posting test Blogs

dhairyashah1, 2022-10-29:
This issue aims to create a separate dynamic webpage that hosts the `documentation` and `README.md`s along with some other features viz. Adding board information, 3D layout + Images, testing blogs. Site hosted on **GitHub Pages**

**Suggestion -** 
Use [jekyll](https://jekyllrb.com/) - a simple site generator using gh-pages. Tutorials can be found [here](https://docs.github.com/en/pages/setting-up-a-github-pages-site-with-jekyll).
Ref - https://sravjti.in/sra-board-component/

**Todo -** 
- [ ] Fix any bugs in the current README.md description files
- [ ] Add/Optimise any technical content
- [ ] Create a different branch called `gh-pages` and then start contributing
- [ ] Host a site including the above information for [Board](https://github.com/SRA-VJTI/sra-board-hardware-design/blob/master/README.md) and Sensors - [LSA](https://github.com/SRA-VJTI/sra-board-hardware-design/blob/master/sensors_2022/lsa/README.md) and [MPU](https://github.com/SRA-VJTI/sra-board-hardware-design/blob/master/sensors_2022/mpu/README.md)
- [ ] Another page or subsection to **display blogs** on debugging/ technical matter observed during the journey of Board development
- [ ] Another page or subsection to **display renders, 3d models, images and renders**. Something like - https://yellow.home-assistant.io/yellow-v1.2-standard/ 

CC - @Dhruv454000 @ChinmayLonkar @Sidshx assign this issue to some SY

SuperChamp234, 2022-10-31:
@dhairyashah1 I can take on this issue.

Looks like there is documentation in the `documentation` folder as well as the main `README.md`. Any other locations that I should include?

Also only document the 2021 SRA Board, right?

dhairyashah1, 2022-11-05:
Yes, apart from the `documentation` and `README.md` files, a custom section for blogs to be created, basically create a folder containing many md files containing the blog content. These can be displayed under the `Blog `section on the web page. 
Additionally, refer last 2 points of the tasklist.

SuperChamp234, 2022-11-07:
Task 1 done, moving on to next tasks

dhairyashah1, 2022-11-20:
Update @SuperChamp234 ?

SuperChamp234, 2022-11-20:
Working on setting up hugo to host the blogs. Sorry for the delay, will make some progress asap.

dhairyashah1, 2022-12-11:
Updates @SuperChamp234

dhairyashah1, 2023-01-10:
@SuperChamp234 post the temporary link of the website created here.

SuperChamp234, 2023-01-11:
Here is the temporary deployment at: https://superchamp234.github.io/sra-board-hardware-design/

A few points to note about the deployment:

1. Currently the Hugo Workflow copies the `README.md`, the LSA README as well as the MPU README, from their folders. If more files need to be added, the workflow will have to be changed(easy to do so).
2. Currently using the `hugo-book` theme.
3. All documentation assets must be placed in `/documentation` folder, as it is statically uploaded.
4. Blog posts can be hosted too, please update about their requirement if needed.

SuperChamp234, 2023-01-11:
I shall now begin working on the display section, or the 6th Todo

dhairyashah1, 2023-02-02:
Update @SuperChamp234

dhairyashah1, 2023-03-11:
@SuperChamp234 updates? Points 5 and 6 are required as well. A section to display specific `blogs` or  discussions and 
#2
> [ ] Another page or subsection to **display renders, 3d models, images and renders**. Something like - https://yellow.home-assistant.io/yellow-v1.2-standard/

this would be desired. Displaying blogs written in `.md` format would be good. One can also look towards showing prominent GitHub issues like #14. Expecting this work to be done ASAP.

CC - @ChinmayLonkar @Premraj02