SRA-VJTI/sra-board-hardware-design #22: Update README.md for SRA_Board_2023, LSA and MPU as per current changes

dhairyashah1, 2023-01-10:
1. The images and references should be complete and within the systematic folders of this repository. eg `assets`.
2. Information should be updated, written properly in points with links to relevant work and in comparison to previous design highlighting `improvements, what and why` incorporated in this year's design.
3. Images, videos, or gifs should be scaled to uniformity.
4. Fixes and improvements with respect to the previous documentation are desired.
5. Links to relevant issues encountered and their solutions can be mentioned as well. I suggest linking this to blog posts later as mentioned in #18 

@ChinmayLonkar @Jamm02 @Premraj02 assign this task to some SY

RajGupta17, 2023-01-13:
I and @rajatkaush11 are doing this work, most of the changes are done, it will be ready in sometime

dhairyashah1, 2023-01-13:
Issue a PR when done. I hope you are working on individual forks of this repo.

RajGupta17, 2023-01-13:
Yes, we have forked this repo and are working on local

RajGupta17, 2023-01-13:
https://github.com/RajGupta17/sra-board-hardware-design

Updated the markdown for SRA Board 2023, the pictures of Motor Driver modes are yet to be updated, rest everything is done according to the suggestions. @dhairyashah1 @ChinmayLonkar @Jamm02

dhairyashah1, 2023-01-16:
@RajGupta17 seems like your fork is quite behind and is not in sync with many of the current upstream commits. Update that asap. Also include latest images of the board

ChinmayLonkar, 2023-03-30:
Readme Updated

closed 2023-03-30 by ChinmayLonkar